aceinnolab/Inkycal #300: inky***.py  crashes when atempting it's second refresh to e-paper display.

km4vig, 2023-12-26:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

km4vig, 2023-12-26:
[inky_output.txt](https://github.com/aceinnolab/Inkycal/files/13772263/inky_output.txt)
[inky_output2.txt](https://github.com/aceinnolab/Inkycal/files/13772264/inky_output2.txt)
[inky_log3.txt](https://github.com/aceinnolab/Inkycal/files/13772265/inky_log3.txt)
[image]

aceisace, 2024-01-05:
Thanks for reporting this issue @km4vig . I have been able to reproduce this error on my end. Thanks to your log outputs, I have been able to find the issue. The cause for this error is not with Inkycal, but with the drivers provided from waveshare, which after a long-due update are not working as expected. The fix is already underway and will be pushed to main soon once I have tested it on my end. 

Please note this issue is the exact same issue as issue #301 and will be closed to avoid duplicated issues. You can continue the conversation and post comments in #301

closed 2024-01-05 by aceisace

aceisace, 2024-01-07:
@km4vig 
Thank you for your patience! The fix I have been testing to fix this problem seems to be working!

Please run the following to fix this issue:
```sh
cd ~/Inkycal
source venv/bin/activate
pip install gpiozero
git pull
sudo reboot
```