aceinnolab/Inkycal #301: gpiozero.exc.DeviceClosed: LED is closed or uninitialized

razem-io, 2023-12-27:
**Describe the bug**
Running python inky_run.py results in an error. Just installed InkycalOS Lite on an `epd_7_in_5_v3_colour` I ordered a few days ago.

```
(venv) inky@inkycal:~/Inkycal $ python inky_run.py
/home/inky/Inkycal/venv/lib/python3.9/site-packages/gpiozero/devices.py:295: PinFactoryFallback: Falling back from lgpio: No module named 'lgpio'
  warnings.warn(
inkycal.modules.inkycal_weather loaded
inkycal.modules.inkycal_calendar loaded
inkycal.modules.inkycal_feeds loaded
26-12-2023 23:24:40 | inkycal.modules.inkycal_slideshow |  ERROR: No images found in the given folder, please double check your path!
No images found in the given folder path :/
loaded inkycal
Inkycal version: v2.0.3

Selected E-paper display: epd_7_in_5_v3_colour
generating image(s) for Weather (openweathermap) - Get weather forecasts from openweathermap...OK!
generating image(s) for Calendar - Show monthly calendar with events from iCalendars...OK!
generating image(s) for RSS / Atom - Display feeds from given RSS/ATOM feeds...OK!
Inkycal version: v2.0.3

Selected E-paper display: epd_7_in_5_v3_colour
Date: 26 Dec 23 | Time: 23:24
Generating images for all modules...successful
Initialising..Updating display......Done
Sending E-Paper to deep sleep...Done

No errors since 1 display updates
program started 47 seconds ago
5 minutes left until next refresh
Date: 26 Dec 23 | Time: 23:31
Generating images for all modules...successful
Initialising..Traceback (most recent call last):
  File "/home/inky/Inkycal/venv/lib/python3.9/site-packages/gpiozero/output_devices.py", line 87, in _write
    self.pin.state = self._value_to_state(value)
AttributeError: 'NoneType' object has no attribute 'state'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/inky/Inkycal/venv/lib/python3.9/site-packages/gpiozero/devices.py", line 580, in _check_open
    super()._check_open()
  File "/home/inky/Inkycal/venv/lib/python3.9/site-packages/gpiozero/devices.py", line 208, in _check_open
    raise DeviceClosed(
gpiozero.exc.DeviceClosed: LED is closed or uninitialized

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/inky/Inkycal/inky_run.py", line 7, in <module>
    asyncio.run(inky.run())  # If there were no issues, you can run Inkycal nonstop
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/inky/Inkycal/inkycal/main.py", line 354, in run
    display.render(im_black, im_colour)
  File "/home/inky/Inkycal/inkycal/display/display.py", line 99, in render
    epaper.init()
  File "/home/inky/Inkycal/inkycal/display/drivers/epd_7_in_5_v3_colour.py", line 77, in init
    if (epdconfig.module_init() != 0):
  File "/home/inky/Inkycal/inkycal/display/drivers/epdconfig.py", line 102, in module_init
    self.GPIO_PWR_PIN.on()
  File "/home/inky/Inkycal/venv/lib/python3.9/site-packages/gpiozero/output_devices.py", line 210, in on
    self._write(True)
  File "/home/inky/Inkycal/venv/lib/python3.9/site-packages/gpiozero/output_devices.py", line 89, in _write
    self._check_open()
  File "/home/inky/Inkycal/venv/lib/python3.9/site-packages/gpiozero/devices.py", line 583, in _check_open
    raise GPIODeviceClosed(str(e))
gpiozero.exc.GPIODeviceClosed: LED is closed or uninitialized
(venv) inky@inkycal:~/Inkycal $
```

zsoltika, 2024-01-02:
I got hit with the same issue :-(
And the display never gets refreshed.

razem-io, 2024-01-02:
This error appears to be very uncommon. My Google search did not result in any valid hits. I will try to invest some more time, but I hope someone with more experience can help.

SIDy1993, 2024-01-03:
I also wondering why it's not a second time refreshing.
Screen epd_7_in_5_v2_colour // V2!
Buyed from tindie bundle

Changed refresh time in settings to 2 minutes to catch the error.

Till it's not fixed I run a crontab every 2 hours with the inky_run script.

One time refresh works and after 2 minutes the script give the error and dies.

Quickfix till his vacation is over I think :)

zsoltika, 2024-01-03:
> I got hit with the same issue :-( And the display never gets refreshed.

OK, the display gets refreshed, I had a typo in the display type config, so I also can run the script from crontab hourly due to the crash.

closed 2024-01-06 by aceisace

aceisace, 2024-01-07:
@zsoltika @razem-io @SIDy1993 
Thank you for your patience! The fix I have been testing to fix this problem seems to be working!

Please run the following to fix this issue:
```sh
cd ~/Inkycal
source venv/bin/activate
pip install gpiozero
git pull
sudo reboot
```

SIDy1993, 2024-01-08:
@aceisace 

Thanks for the fix.
Yesterday after you closed it I ran git pull and it already works after it with the little update of 1 file I think.

Now I tested your comment but it seems everything alright already

`Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: gpiozero in ./venv/lib/python3.9/site-packages (2.0)
Requirement already satisfied: colorzero in ./venv/lib/python3.9/site-packages (from gpiozero) (2.0)
Requirement already satisfied: setuptools in ./venv/lib/python3.9/site-packages (from colorzero->gpiozero) (69.0.2)`

aceisace, 2024-01-08:
Glad to hear it works now! Yes, I did push a fix but I wanted to test it a little longer on my Raspberry Pi to ensure it would ran stable even after several updates. Now that I have tested it myself and it's working as expected, the fix is official 😉

zsoltika, 2024-01-08:
> @zsoltika @razem-io @SIDy1993 Thank you for your patience! The fix I have been testing to fix this problem seems to be working!
> 
> Please run the following to fix this issue:
> 
> ```shell
> cd ~/Inkycal
> source venv/bin/activate
> pip install gpiozero
> git pull
> sudo reboot
> ```

I did it but `pip install ...` have not installed anything (like for @SIDy1993 : requirements already satisfied). Shouldn't it be `pip install gpiozero --upgrade` ?

aceisace, 2024-01-08:
@zsoltika If pip install doesn't install anything extra, it means you already have the dependencies installed. As Inkycal uses fixed versions for most dependencies, there is no need to upgrade as some things may break otherwise.
Basically, gpiozero should be present in recent versions of Inkycal, but not older ones, hence I added it anyway so older users can benefit from this too. 👍 
Is your Inkycal running correctly after the fix?

zsoltika, 2024-01-09:
@aceisace Thanks for the clarification! 
Yes mine works as expected since the update thanks for the fix as well!