aceinnolab/Inkycal #305: [FEATURE]: Add language "Austrian"

the-sim87, 2024-01-12:
Feature request for language "Austrian": Here in austria, it's called "Jänner" instead of "Januar".
Everything else should be the same as "German"

the-sim87, 2024-01-12:
Hmm... i had a quick view in the code, i think the names from the months are coming from the calender-url, right? Then it's not so easy as I thought...

aceisace, 2024-01-13:
Thanks for the feature suggestion @the-sim87 . The translation is automatically generated via the arrow dependency, which is responsible for timezone and localisation support in Inkycal. As a result, adding languages is pretty easy.

You can either set the language from `"de"` to `"de-at"` in your `settings.json` file or use the web-ui to select Austrian (support added) and generate a new settings.json file.

I'm closing this issue as the feature has been implemented 👍

closed 2024-01-13 by aceisace