aceinnolab/Inkycal #311: [FEATURE]: Pi-Sugar support

aceisace, 2024-01-16:
The feature request this time is about a battery-powered Inkycal with the help of a PiSugar pcb board. 
This PCB has a built-in RTC (real-time-clock) and can be used to make Inkycal even more efficient by shutting it down when it's not being used. 

The PiSugar board is available from [here](https://www.tindie.com/products/pisugar/pisugar-3-battery-for-raspberry-pi-zero/) for around 40 usd and is not associated with Inkycal in any way.

With the PiSugar board, it should be possible to extend the battery life of Inkycal by a few days at least. However, exact measurements still need to be conducted.

aceisace, 2024-01-24:
Status update: PiSugar has been ordered and is now on the way. Once I have done some testing, I will post an update on this.
This has been funded by the Inkycal buyers on Tindie and all the sponsors of InkycalOS-Lite. Thank you very much for your contribution!

J-CMartin, 2024-04-22:
Any progress on this?

aceisace, 2024-04-23:
Hi @J-CMartin and thanks for your interest.
Due to some circumstances, I had to prioritise some other topics, including the development of the inkycal-embedded-hardware system, which took a significant amount of time recently. 
This feature is definitely on my todo and I had already started development on this including the integrated web-ui for managing Inkycal and the Rapsberry Pi (for reboots, logs etc.), I will take care of this feature in the next two weeks.

aceisace, 2024-05-11:
The changes so far boil down to the following:
- [ ] Implementing a simple file-like database (json) 
- [ ] Adding time-ranges to the web-ui for the peak-timerange and off-peak timerange (in which a calibration may be carried out)
- [ ] Saving the state of Inkycal after each cycle
- [ ] Extend the existing code to allow running Inkycal just once, then stop. After the stop, a shutdown may be initiated.

Further improvements:
- Drop top_level variable in favour of dynamic variables via the new settings file for inkycal-internal operations to set cache-folders, image-folders and even VCOM for parallel displays.
- refactor parallel display drivers to use vcom from said settings file
- Make inkycal_slideshow remember the index of the current position via the json-cache
- minor improvements to countdown 
- separate logging from main.py

J-CMartin, 2024-05-16:
**Suggestion**: I didn't see this (Sorry if I overlooked it), but we will definitely need a battery indicator to display on one of the corners so we can visually see when the battery needs to be charged. 
This could be a small overlay on top of other modules, it would cover a small rectangle in one corner

aceisace, 2024-05-18:
This sounds like a good idea, but do you have a sample code on how to read the voltage from the sugarpi? From the docs over [here](https://github.com/PiSugar/PiSugar/wiki/PiSugar-3-Series), it seems one needs to access the i2c registers, but it's not really specified how

aceisace, 2024-05-18:
Edit: Looks I found something, but this is from the shell only:
```
sudo i2cdump -y 1 <hex_value>

Battery capacity reading
Register address:
0X22, 0X23, 0X2A
Register operation:
can be read
Data description:
0x22 is high battery voltage, 0x23 is low battery voltage. When reading, you should first arrange the two bits of data in order to get the actual voltage in mA.
For example, if the data read are 0x10 and 0x1F, the combined result will be 0x101F=4127mV=4.127V.
The 0X2A address is the calculated battery power percentage.
Function description:
WebUI can read the battery power percentage
```

J-CMartin, 2024-05-18:
Once the power manager software is installed you can use the API to interface with it. 
https://github.com/PiSugar/PiSugar/wiki/PiSugar2#api
No need to drive into I2C world 🙂

aceisace, 2024-05-18:
Didn't know that, thanks very much for the info!

closed 2024-06-25 by aceisace