aceinnolab/Inkycal #321: The build fails

greg19, 2024-02-12:
**Describe the bug**
The build fails.

**To Reproduce**
Install the Raspberry Pi OS 32 bit on Raspberry Pi Zero. Update the system with `sudo apt update && sudo apt upgrade`. Follow the [Configuring the Raspberry Pi](https://github.com/aceinnolab/Inkycal?tab=readme-ov-file#configuring-the-raspberry-pi) and [Installing Inkycal](https://github.com/aceinnolab/Inkycal?tab=readme-ov-file#installing-inkycal) guides on Raspberry Pi Zero.

**Expected behavior**
The build completes successfully.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Raspberry Pi OS 32 bit
 - Version: 2023-12-5

aceisace, 2024-02-12:
Hi there and thanks for opening this issue @greg19 
I'm sorry to hear the installation isn't going well right now. In fact, I'm updating dependencies as we speak since they needed an update, hence some things are breaking right now. I'll report back to you once I've finished testing and the pipeline is working again 👍

aceisace, 2024-02-13:
Hi @greg19 , I'm coming back to you to let you know that the installation should now work fine.
Please let me know if the issue persists and close the issues if the issue has been resolved.

aceisace, 2024-02-13:
You can jumpt to the fixed version with the following:
```sh
cd ~/Inkycal
source venv/bin/activate
git reset --hard
git pull
pip install -r requirements.txt

# test if it works:
python inky_run.py
```

greg19, 2024-02-13:
It works now, thanks!

By the way, it would be awesome to have something like a `stable` branch that contains newest working version of the project.

closed 2024-02-13 by greg19