aceinnolab/Inkycal #322: [FEATURE]: Request that Inkycal support update intervals that are greater than 60

emilyboda, 2024-03-04:
**Problem**
Setting the update interval to a number greater than 60 (such as 720) results in an error:
`IndexError: list index out of range`

**Feature request**
- Support update intervals greater than 60. Some uses of Inkycal may not need update intervals greater than one day
- If update interval is set to 1440 (one day) have Inkycal update at 12:01am since this is likely displaying something like a calendar that needs to be updated only once after midnight.

**Full Error so that others may google and find this**
```
Traceback (most recent call last):
  File "/home/pi/Inkycal/inky_run.py", line 7, in <module>
    asyncio.run(inky.run())  # If there were no issues, you can run Inkycal nonstop
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/pi/Inkycal/inkycal/main.py", line 372, in run
    sleep_time = self.countdown()
  File "/home/pi/Inkycal/inkycal/main.py", line 174, in countdown
    minutes = [_ for _ in update_timings if _ >= now.minute][0] - now.minute
IndexError: list index out of range
```

aceisace, 2024-03-10:
@emilyboda This has been implemented now and is open for testing. Although the unittests are passing, please let me know if it works as expected once you're able to test this 👍

closed 2024-03-10 by aceisace

emilyboda, 2024-03-24:
Running with no issues for a week!

aceisace, 2024-03-24:
Awesome! Glad to hear that and thanks for the info @emilyboda !