aceinnolab/Inkycal #326: [Bug]: Red circle and frame off center

mygrexit, 2024-03-17:
**Describe the bug**

This issue is regarding the calendar module:

The red circle around the current day and the border lines around days with events are off center.

I can only speak for the 12.48 waveshare display.

**To Reproduce**
Steps to reproduce the behavior:

Install latest verision.

Expected behavior

As before. Centered frame and centered 
[image]
red dot.

**Screenshots**

Photo is from @SIDy1993

KTuffyW623, 2024-06-21:
I had the same issue and could fix it by modifying a few lines in "inkycal/modules/inkycal_calendar.py":

1.

```
text_bbox_height = self.font.getbbox("hg")
weekdays_height = int((text_bbox_height[3] - text_bbox_height[1])* 1.25)

```
->
```
weekdays_height = sum(self.font.getmetrics())
```

2. 

Every occurance of

```
text_bbox_height = self.font.getbbox("hg")
line_height = text_bbox_height[3] + line_spacing

```
->
```
text_bbox_height = sum(self.font.getmetrics())
line_height = text_bbox_height + line_spacing
```

I changed some lines in "inkycal/custom/functions.py" in the same fashion as in 2.

That helped me to center everything.

aceisace, 2024-06-22:
Apologies for getting back so late at this. I've got the fix ready for this, will create a PR for this now and will attempt to merge it once the tests are passing

closed 2024-06-22 by aceisace