aceinnolab/Inkycal #327: Inkycal does not work with latest linux kernel update (6.6.20)

the-sim87, 2024-03-22:
**Describe the bug**
Since the latest kernel update to 6.6.20 inkycal stopped working, I get a GPIO error. There is a change in GPIO of the kernel:
https://forums.raspberrypi.com/viewtopic.php?t=361116&start=100#p2203689

**To Reproduce**
sudo apt-get dist-upgrade 
reboot

Raspberry Pi Zero 2:
 - OS: Raspberry Pi OS Lite 64 Bit, Debian 12 (Bookworm)

aceisace, 2024-03-23:
I've pushed the required changes in the `hotfix/327` branch. May I kindly ask you to run Inkycal via that branch _with_ the latest kernel? Once you can confirm it works, I will merge it into main.
```sh
cd ~/Inkycal
git pull
git checkout hotfix/327
source venv/bin/activate
python inky_run.py
```

This assumes you have SPI enabled in the latest kernel. Please use the latest kernel for this test.

the-sim87, 2024-03-23:
Hi!
Does not work:
`user@pi-display:~/Inkycal $ cd ~/Inkycal
user@pi-display:~/Inkycal $ git pull
Already up to date.
user@pi-display:~/Inkycal $ git checkout hotfix/327
error: pathspec 'hotfix/327' did not match any file(s) known to git
user@pi-display:~/Inkycal $
`

aceisace, 2024-04-03:
I've adapted the README instructions so that new users do not attempt to use the latest version. With a link to the last supported version of Raspberry Pi OS and a warning that the latest version will not work, this should be complete, at least for now. Thanks for reporting this @the-sim87 !