aceinnolab/Inkycal #331: Unable to determine board revision from /proc/cpuinfo

jo9778, 2024-04-09:
I tried a few times to get the Inkycal up and running on a RPI 4B as described, but i always get this error message. How can this be fixed or what i´m doing wrong? I´m using the 12.48 inch waveshare 3 colour Display with RPI 4B. The used Raspian Image is the linked one from 11.12.2023 in the description.

[image]

Maybe someone had the same issue and can help me?

Thank you

jo9778, 2024-04-10:
Thank you for your fast reply!
The Pi Image i used, is the linked one in your answer. To double check, i´ve downloaded it again and tried the complete installation again. After finishing, the "git pull" command response is "Already up to date.".
[image]

Are there any other possibilities?

aceisace, 2024-04-12:
You're welcome 😄 
I've looked into this issue a little deeper and it seems that this is related to a part of the driver for which only the compiled version is available from waveshare i.e. it is not possible to read out the code. This issue should therefore not be reported on Inkycal as the driver section of Inkycal is actually using the waveshare drivers with little to no modifications.

While spotting for differences between the latest available version of the drivers, I have seen that the epdconfig file for the 12.48" display has been updated. I've therefore updated this too, in hope that it might fix the error you are getting, but I haven't yet had the time to test it out yet.

Therefore, please run `git pull` again and attempt to run Inkycal as you did before. If you get the same error, please report this issue here: https://github.com/waveshareteam/12.48inch-e-paper/issues

If you get a different error than the current one, chances are I've overlooked something and that should be reported here with the full traceback

jo9778, 2024-04-14:
Thank you very much for your great work. After running git pull the display is starting up and shows the calendar as expected. There are no more errors reported. 
After the first try with the calendar module, i tried to add another modules to display, but neither the "Show preview" on the online GUI nor the real display shows more than the first module. There were also no Errors displayed on the debug line on the bottom. Are there any further steps necessary to display more than one module on a page?

aceisace, 2024-04-14:
Glad to hear it's working now 😄 
Please note that the show-preview function of the web-ui is still experimental i.e. it will not update the content nor will the settings.json file be updated after editing something in the web-ui after the first submit/show preview click. If there was no indication that there was an error, chances are the settings.json file did not reflect the changes of the web-ui after the first submit.

As this is a separate issue and since this issue is resolved, I will close this issue and open another one for the web-ui. In the meantime, you can head to the web-ui, add all details as needed and then click on generate or preview.

closed 2024-04-14 by aceisace