aceinnolab/Inkycal #334: Display 12,48 inch not updating

jo9778, 2024-04-23:
That bug is in adition to the problem
"Unable to determine board revision from /proc/cpuinfo" 
about one week ago:

After powering up the system , initialisation of the display is done and things were displayed as expected. The ssh console says - "Updating done - Sending display to deepsleep mode"

And then next refresh in xy minutes (as expected). After this time, the system starts up again and tries to refresh but gets stuck in this situation (see screenshot). Interestingly, 9 out of 10 times there is also no different situation after power cycling the whole system.
Maybe there is a little driver or I/O Problem left?

[image]

The screenshot is from the first startup after the installation process. After power cycling, the system gets stuck within the line "Generating images for all modules...Initialising..EPD init..."

Thank you

aceisace, 2024-04-29:
This is very likely related to #335 and should therefore be fixed with the latest commit. Please update Inkycal and try again:
```sh
cd ~/Inkycal
git pull
# to test if it works:
source venv/bin/activate
python inky_run.py
```

aceisace, 2024-05-11:
@jo9778 Can you confirm that it works please so that I can close this issue? Thanks in advance!

jo9778, 2024-05-30:
Sorry for my late answer. After updating i got the same issue as before. It´s  described here (#331). I decided to replace the RPi4 by a Rpi zero. With this setup, inkycal is working perfect.

aceisace, 2024-05-31:
Thanks for getting back @jo9778 
In that case, I'll be closing this issue and the other issue can be discussed at #331 👍

closed 2024-05-31 by aceisace