aceinnolab/Inkycal #335: 12.48 screen stuck on EPD Init....

J-CMartin, 2024-04-25:
**Describe the bug**
After a fresh install on a new Pi4, the display won't update. I have a Pi02W that works fine with that display so I know the display is working correctly. 
76b86afafb8bd234642184a406c00abe5af7598d

**To Reproduce**
Installed Inkycal on a Raspberry Pi4 following the steps on the ReadMe section
Configured settings.json and run InkyCal, Image is generated on Image folder but Screen is not updated. 
Cosole shows "EPD Init..." and stays there indefinitely.

**Expected behavior**
Screen should be updated with the image generated and "Done" message should be displayed on console. Alternatively an error message should display to help identify the problem. 

**Additional context**
After comparing the code between the 2 Pi's I realized that my Pi0 had a slighter older version of Inkycal and after some research I found out that a commit was applied to Inkycal specifically for the 12.48 display. After comparing the affected files and reversing the changes on that commit the Pi4 started working again. I suggest revert the code to the pre-commit version unless someone can explain how they made it work with the 12.48 screens. 

Couple of Notes regarding this:
1. WiringPi was picked up by a different developer and it's being actively developed again, here are the instructions to install the latest version (3.2 as of the creation of this Issue):
```
git clone https://github.com/WiringPi/WiringPi
cd WiringPi
./build
```
I suggest that this is added after the "sudo apt-get install" and before installing Inkycal (It needs git to be installed on the system) 

2. It feels like this was broken by waveshare when they added code to support Pi5 without proper regression testing. I found a commit on waveshare where they have the support for both 32 and 64 bits OS' but it still works with Pi4. I'm test this one more time and let you know where to get the right libraries from there.

J-CMartin, 2024-04-26:
In order to maintain compatibility with both 32 and 64 bit OS', here what you can do:
Go to the Waveshare commit previous to latest Pi5 commit here:
https://github.com/waveshareteam/12.48inch-e-paper/tree/6eca3c55e264d974eb89ae4fcddaa37ea2a9775c/RaspberryPi/python/lib

From that waveshare folder, use the following files and rename them according to your files already on inkycal/display/drivers/epdconfig_12_in_48.py:
epdconfig.py  -->  epdconfig_12_in_48.py
DEV_Config_64.so --> epd_12_in_48_lib_64bit.so
DEV_Config_32.so --> epd_12_in_48_lib_32bit.so

J-CMartin, 2024-04-26:
I also opened an Issue on waveshare repository to let them know that their latest code is not working as expected.

aceisace, 2024-04-28:
linked to #334 . Working on this now

aceisace, 2024-04-28:
@J-CMartin Thanks for finding the bug, providing detailed information on how to fix this and for testing!

Could you please take a look at the PR #336 and provide your input if the changes are looking ok? Thank you in advance!

J-CMartin, 2024-04-28:
I downloaded hotfix#335 and it worked as expected on my Pi4. Let me test it on my Pi0 as well just in case

J-CMartin, 2024-04-29:
Test was successful with the Pi Zero2W as well. It looks good. Please see my comments on the READ.ME file

aceisace, 2024-04-29:
Thanks for testing and letting me know @J-CMartin !
I'll adjust the README accordingly before merging  👍

closed 2024-04-29 by aceisace