aceinnolab/Inkycal #337: Driver Issue for 7.5" Display due to Kernel update

tdortton, 2024-05-07:
**Describe the bug**
Unable to start Inkycal with inky_run command, resulting in a RunTime Error. Ace thinks it's a driver issue due to Raspberry Pi OS Kernel update.

**Attachments**
Console Output of Error log
[ConsoleOutput.txt](https://github.com/aceinnolab/Inkycal/files/15235242/ConsoleOutput.txt)

**Setup (please complete the following information):**
 - OS: Raspberry Pi OS
 - Raspberry Pi Zero 2W
 - Waveshare 7.5" 3 Color E-Ink display

aceisace, 2024-07-08:
Hi there, @tdortton . May I kindly ask you to test this according to the instructions from the README, using the specified release of Raspberry Pi OS. It doesn't seem this is related to Inkycal itself, hence this issue should be reported on the Waveshare GitHub repo

aceisace, 2024-11-07:
Closing this issue since the mentioned error still has no fix other than using the previous version of Raspberry Pi OS as mentioned in the README. Waveshare also has yet to provide an official fix for this.

Further resources for those interested in this issue:
* https://github.com/waveshareteam/e-Paper/issues/350
* https://github.com/waveshareteam/e-Paper/issues/336

closed 2024-11-07 by aceisace