aceinnolab/Inkycal #340: Translations are not 100%

EinKantHolz, 2024-06-11:
Just want to know if it's a bug or a missing "feature": I set all my modules to "de" language but there are some words which are not translated.
- The Weater module: Every day is in english
- The Agenda module: for all day entries it says "all day" instead of "ganztags" or similar in german.

Can i modifiy those translations by myself and if so, where?

aceisace, 2024-06-11:
Hi @EinKantHolz and thanks for reporting this issue. I will take care of this soon 👍 . Meanwhile, if you notice any other missing translations, please post them here

aceisace, 2024-06-22:
I've implemented the suggestions by these changes:
- Add MaterialIcons font
- Use icon for all-day events on Agenda module as translating "all-day" is pretty complicated without bloating the code and additional dependencies
- Fix missing translations, e.g. weekday on daily weather forecast

The PR for this is on the way, once the tests have passed, I kindly ask you to test the changes to ensure it's working as expected

closed 2024-06-22 by aceisace