aceinnolab/Inkycal #342: All-day-Appointments not shown all days in agenda

EinKantHolz, 2024-06-18:
**Describe the bug**
While using the Agenda module, i found a bug. All-day-appointments that are going more than one day are not shown on "all days" but only the first day (point of start). I've attached a photo of the screen and a screenshot of the calendars on the server for comparison.

Calendars on server:
[image]

Agenda module:
[image]

aceisace, 2024-06-18:
@EinKantHolz Hi there and thanks for reporting this issue! 
Yes, indeed, I can confirm this is a bug by looking at the provided image. I'll take care of this soon 👍

EinKantHolz, 2024-08-18:
Okay so now the multi-day-events are pinned on top. Is that correct?
May I suggest adding the start and end date? That would be more helpful. But over all it's working.

aceisace, 2024-08-22:
@EinKantHolz Yes, you're correct. Can you show me the rendering of your agenda module as I don't use the agenda module on my build? Dates can be added too, however, depending on the layout, there may not be enough space (i.e. 5.3" e-paper display). In the future, this will be solved with widgets, however, an image would be great to help understand the issue. 

This issue, i.e. multi-day events not spanning across several days should be resolved right?

EinKantHolz, 2024-08-25:
yeah sure. There are several issues here. Let me explain step by step.
First you can see the agenda module with 3 events:

- all-day event in the past
- A single all-day event
- all-day event over more days

[image]

Here are the bugs/problems:

**1. all-day event in the past**
This event started on August 24th and is ending on August 27th. You can see, that this event is "pinned" at the top but is still missing on the other days. Thats okay (the starting point is in the past) but it would be nice if there is the starting and ending date attached somewhere (in the title or maybe a second smaller line). Or expand and show the event for all days acutally. 

**2. A single all-day event**
This is correct. It is only for the 25th of August and is shown as it should be. The event will disappear after the date is in the past and not longer shown in the agenda.

**3. all-day event over more days**
The second bug/problem: This event starts today (August 25th) and will end on August 27th but is not shown for the other days. It's similar to the first problem. And if the starting date is in the past, this "bug" will become like #1.

[image]

I'm not sure how big this issue is concerning the code and architecture. My preffered solution would be to display the cross-day events for each day but you already mentioned, that this isn't so easy, as far as I can remember.
So maybe a second smaller line (or after the title) for all cross-day events like (25.08.2024 - 27.08.2024) and left if pinned at the top for starting date in the past.

closed 2026-10-20 by github-actions[bot]