aceinnolab/Inkycal #347: InkyCal 2.03 stops refreshing on Pi Zero W, Python using 80% cpu

mattenz, 2024-06-23:
**Describe the bug**
After InkyCal has been running for a bit (sometimes hours, sometimes days), it seems to just stop refreshing after a while, however it looks like the Python process is still running under this scenario and consuming a lot of CPU (80-85%). At the moment I have to schedule reboots of the Pi to keep InkyCal refreshing reliably.

**Screenshots**
The logs of InkyCal show that its last recorded refresh was an hour later than what is shown on the display itself (in this case, the logs say 21:01 but the display shows 20:01 - no more log events are logged from that time tho):
`23-06-2024 20:01:00 | inkycal.modules.inkycal_weather |  INFO: Image size: (476, 100)
23-06-2024 20:01:02 | inkycal.modules.inkycal_weather |  INFO: Connection test passed
23-06-2024 20:01:09 | inkycal.modules.inkycal_calendar |  INFO: Image size: (460, 444)
23-06-2024 20:01:11 | inkycal.modules.ical_parser |  INFO: loaded iCalendars from URLs
23-06-2024 20:01:12 | inkycal.modules.inkycal_feeds |  INFO: Image size: (460, 158)
23-06-2024 20:01:13 | inkycal.modules.inkycal_feeds |  INFO: Connection test passed
23-06-2024 20:01:14 | inkycal.main |  INFO: successful
23-06-2024 20:01:16 | inkycal.modules.inky_image |  INFO: merged given images into one
23-06-2024 21:01:00 | inkycal.modules.inkycal_weather |  INFO: Image size: (476, 100)
23-06-2024 21:01:02 | inkycal.modules.inkycal_weather |  INFO: Connection test passed
23-06-2024 21:01:12 | inkycal.modules.inkycal_calendar |  INFO: Image size: (460, 444)
23-06-2024 21:01:14 | inkycal.modules.ical_parser |  INFO: loaded iCalendars from URLs
23-06-2024 21:01:15 | inkycal.modules.inkycal_feeds |  INFO: Image size: (460, 158)
23-06-2024 21:01:16 | inkycal.modules.inkycal_feeds |  INFO: Connection test passed
23-06-2024 21:01:17 | inkycal.main |  INFO: successful
23-06-2024 21:01:19 | inkycal.modules.inky_image |  INFO: merged given images into one`

**Additional context**
The Pi is running Bullseye. My config file is as below:
`{
    "model": "epd_7_in_5_v2",
    "update_interval": 60,
    "orientation": 0,
    "info_section": true,
    "info_section_height": 30,
    "border_around_modules": false,
    "calibration_hours": [
        0,
        12,
        18
    ],
    "modules": [
        {
            "position": 1,
            "name": "Weather",
            "config": {
                "size": [
                    480,
                    120
                ],
                "api_key": "api key",
                "location": "2191913",
                "round_temperature": true,
                "round_windspeed": true,
                "forecast_interval": "daily",
                "units": "metric",
                "hour_format": "24",
                "use_beaufort": true,
                "padding_x": 2,
                "padding_y": 10,
                "fontsize": 12,
                "language": "en"
            }
        },
        {
            "position": 2,
            "name": "Calendar",
            "config": {
                "size": [
                    480,
                    464
                ],
                "week_starts_on": "Monday",
                "show_events": true,
                "ical_urls": "http://outlook.office365.com/owa/calendar/<id>/calendar.ics",
                "ical_files": null,
                "date_format": "DMMM",
                "time_format": "HH:mm",
                "padding_x": 10,
                "padding_y": 10,
                "fontsize": 12,
                "language": "en"
            }
        },
        {
            "position": 3,
            "name": "Feeds",
            "config": {
                "size": [
                    480,
                    178
                ],
                "feed_urls": "http://blackpearl.fortress.local/rnzcustom.xml",
                "shuffle_feeds": true,
                "padding_x": 10,
                "padding_y": 10,
                "fontsize": 12,
                "language": "en"
            }
        }
    ]
}`

I can't see any other instances running:
`ps -aux | grep python
matt       255 40.4 18.2 174936 80352 ?        Rl   17:20 160:44 venv/bin/python inky_run.py
matt       739  0.0  3.8  52356 16884 ?        S    17:21   0:04 /usr/bin/python3 /usr/share/system-config-printer/applet.py
matt      1410  0.0  0.4   7432  1900 pts/1    S+   23:58   0:00 grep --color=auto python`

aceisace, 2024-06-23:
Hi @mattenz and thanks for opening this issue.

I'm using the latest version of Inkycal (from yesterday) and even on my Zero W, I do not see any cpu-intensive task running there as the total load is less than 2% on average (see screenshot below);

[image]

Can you please check the following:
- Ensure `crontab -e` contains only one line
- Ensure only one instance of Inkycal is running
- Have you installed something else apart from Inkycal?
- Ensure you are using the latest version of Inkycal
- Ensure that your Raspberry Pi is getting a good wi-fi signal

My `settings.json` file is not that different from yours, so I believe it is related to something else. My build is also based on bullseye, hence it should be possible to reproduce the bug in some way

mattenz, 2024-06-24:
I think I'm using the latest version...currently 2.0.3. This has been happening a while though on a previous version and I thought updating it a few weeks back to 2.0.3 might solve it but unfortunately not.

To answer your quesitons:

- Only one crontab-e entry
- Only one instance as far as I can see (there's only one Python entry in Top at least)
- Inkycal is the only extra thing installed on it
- 2.0.3
- Yes

closed 2024-11-13 by github-actions[bot]