aceinnolab/Inkycal #348: ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory

naeldavid, 2024-06-24:
So I have a raspberry pi and while scrolling on github, I found Inkycal. For me it was happiness at first sight. Finally something to do with my raspberry pi ! I flashed my micro sd card with a brand new Raspi OS and followed all instructions. While running the code `python3 inky_run.py` I got this Import error : `ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory`

I made sure everything from the requirements.txt file was installed, even started off from scratch twice but I still get the issue. Any Help ?

[image]

 - OS: Raspberry OS Lite
 - Latest Version

Any module I should install ? Or is it a directory/file error and I need to re-install ?

aceisace, 2024-06-24:
@naeldavid Hi there,
thanks for informing me about this. Are you sure you ran the following command from the README:
```sh
sudo apt-get install git zlib1g libjpeg-dev libatlas-base-dev rustc libopenjp2-7 python-dev-is-python3 scons libssl-dev python3-venv python3-pip git libfreetype6-dev wkhtmltopdf libopenblas-dev

```

This issue would be new to me as the tests on Raspberry Pi OS are passing. Which OS and version are you using?

naeldavid, 2024-06-25:
Hey @aceisace,
I'm sure I ran the command, did it 2 two times to make sure everything works but still get the issue. I have a Raspberry Pi Zero with Bookworm Running :
[image]

naeldavid, 2024-06-25:
I re-flashed the micro-sd card and i'm reconfiguring the Pi. Let you know if anything happens this time.

naeldavid, 2024-06-25:
I also have an idea that could, in my opinion make Inkycal **world famous**. If we could get in touch to talk about it, it would be great.

aceisace, 2024-06-25:
I'll try to reproduce this myself on bookworm, though currently I believe that I am using bullseye. However, this will still have to wait at least a few days, as I currently only have access to my own Inkycal. At the moment, this is being used for testing PiSugar integration and stability according to #311 . 

I cannot rule out the possibility that it is related to python3.11 , yet, this appears to be a known issue related to dependencies of Pillow and ought to be easy to fix: https://stackoverflow.com/questions/48012582/pillow-libopenjp2-so-7-cannot-open-shared-object-file-no-such-file-or-directo

I'm open to all new suggestions and improvement ideas and there is also a lot of work happening behind the scenes too. You can get in touch with me via Discord on the official Support server

naeldavid, 2024-06-25:
While following instruction on the link you sent I got some new errors but I think I know the solution. Anyways, I do not have discord so I cant contact you from there.

aceisace, 2024-06-25:
What was the solution for this issue? It would be helpful if someone else stumbled across the same issue to find out what went wrong and how to fix it.

Interesting... I actually already have a fully working prototype for this idea 😄 
I wrote my reverse-engineered mawaqit-parser which does not require api-keys or an account for my city. My prototype can hold 4000 updates before requiring a recharge, but does not support audio as it's more complex and will significantly reduce battery-life. 

All good then, you can contact me via aceisace63@yahoo.com . Please give an ok so I can delete this message. Not good sharing details on open channels for longer than needed after all

naeldavid, 2024-06-25:
Thank You for the email and I think I found the solution : In the Manual Installation documentation you're asking to download python libraries with `sudo apt-get install`. Install with pip and the person should be good. 

[image]

I saved it as Ace Github if thats ok with you :)

aceisace, 2024-07-08:
Closing this issue as the problem has been resolved