aceinnolab/Inkycal #361: [FEATURE]: Shell-usage

Stopuhr, 2024-07-08:
It would be nice, if one could start inkycal with arguments right from the shell. 
This way, when not running in default mode, no changes have to be done to python-script, that have to be redone after updates etc.
Usage in crontab maybe easier. Looking at the code, I saw some arguments, that have to be passed explicitly through the python-scripts.

aceisace, 2024-11-07:
@Stopuhr Hi there. I have some time to work on this today and wanted to ask you more about this feature request.

Generally, a purely argument-based command-line-tool makes it a bit difficult for users who do not know much about Terminal etc., especially since there are a lot of displays to choose from and the correct one has to be typed exactly for it to work. You could add help for each argument, but this would quickly bloat the script size and complexity and still not be as user-friendly.

The solution I want to implement is similar to the `raspi-config` tool available from the command-line (terminal gui or tui). You can then select a display by using the top or bottom arrows from the list of displays which is fetched directly from the Inkycal folder each time. For starters, I'm implementing some easier operations, like showing a demo image or calibrating the display. However, once this is implemented, I could implement a few more features in this tool.

Could you elaborate more on this part:
_no changes have to be done to python-script, that have to be redone after updates etc.
Usage in crontab maybe easier. Looking at the code, I saw some arguments, that have to be passed explicitly through the python-scripts._

Thanks in advance!

aceisace, 2026-06-30:
Implemented in the latest release