aceinnolab/Inkycal #365: Requires current OS "bookworm," not "buster"

wenestvedt, 2024-07-09:
**Describe the bug**
The current version of Inkycal, 2.0.4, wants the library `arrow` version 1.3.0. But that version of the `arrow` library dropped support for Python 3.7 & 3.8 -- and the previous release of Raspberry OS is base on Debian "buster," and had Python 3.7.

**To Reproduce**
Steps to reproduce the behavior:
On a "buster" host, follow the installation instructions on the Github page. When executing the command `pip install -e` the following error appears:
```(venv) pi@raspberrypi:~/Inkycal $ pip install -e ./
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining file:///home/pi/Inkycal
  Preparing metadata (setup.py) ... done
Collecting PyYAML==6.0.1 (from inkycal==2.0.4)
  Using cached https://www.piwheels.org/simple/pyyaml/PyYAML-6.0.1-cp37-cp37m-linux_armv6l.whl (45 kB)
Collecting appdirs==1.4.4 (from inkycal==2.0.4)
  Using cached https://www.piwheels.org/simple/appdirs/appdirs-1.4.4-py2.py3-none-any.whl (9.6 kB)
INFO: pip is looking at multiple versions of inkycal to determine which version is compatible with other requirements. This could take a while.
ERROR: Ignored the following versions that require a different python version: 1.3.0 Requires-Python >=3.8
ERROR: Could not find a version that satisfies the requirement arrow==1.3.0 (from inkycal) (from versions: 0.1, 0.1.1, 0.1.2, 0.1.3, 0.1.4, 0.1.5, 0.1.6, 0.2.0, 0.2.1, 0.3.0, 0.3.1, 0.3.2, 0.3.3, 0.3.4, 0.3.5, 0.4.0, 0.4.1, 0.4.2, 0.4.3, 0.4.4, 0.4.6, 0.5.0, 0.5.4, 0.6.0, 0.7.0, 0.8.0, 0.9.0, 0.10.0, 0.11.0, 0.12.0, 0.12.1, 0.13.0, 0.13.1, 0.13.2, 0.14.0, 0.14.1, 0.14.2, 0.14.3, 0.14.4, 0.14.5, 0.14.6, 0.14.7, 0.15.0, 0.15.1, 0.15.2, 0.15.3, 0.15.4, 0.15.5, 0.15.6, 0.15.7, 0.15.8, 0.16.0, 0.17.0, 1.0.0, 1.0.1, 1.0.2, 1.0.3, 1.1.0, 1.1.1, 1.2.0, 1.2.1, 1.2.2, 1.2.3)
ERROR: No matching distribution found for arrow==1.3.0
(venv) pi@raspberrypi:~/Inkycal $ 
```

**Expected behavior**
The installation should conclude successfully.

**OS**
```pi@raspberrypi:/etc $ cat os-release 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

**Additional context**
Is it possible to run this on "buster," or should the requirements (and my Pi!) be updated to use "bookworm"?

aceisace, 2024-07-10:
Hi there and thanks for reporting this issue, @wenestvedt . Actually, bookworm was the previous release of Raspberry Pi OS, with Python3.9 as the default version. Nowadays, bookworm is the latest release with python 3.11. Support for Python 3.8 and lower has been dropped in the latest version of Inkycal as it gets increasingly difficult to maintain the code while keeping backward-compatibility in mind.

I'd recommend using this specific version of Raspberry Pi OS: https://downloads.raspberrypi.com/raspios_lite_armhf/images/raspios_lite_armhf-2023-12-11/2023-12-11-raspios-bookworm-armhf-lite.img.xz 
as I can personally confirm this to be working with Inkycal. Alternatively, you can also sponsor Inkycal once to get access to InkycalOS-Lite, which is the ready-to-flash version of Inkycal v2.0.4 which doesn't require any installation or setup.

wenestvedt, 2024-07-10:
Good grief, you're right: I _am_ two releases behind. InkyCal has been running so well, for so long, that I never touched this Pi!

I will take care of the upgrade ASAP. Thanks for the explanation!

closed 2024-07-10 by wenestvedt

aceisace, 2024-07-10:
I'm happy to have been of help and to hear that Inkycal has been running stable for a long time at your end 😊
Sure thing, if you need any additional assistance, please do not hesitate to open an issue here or via Discord

wenestvedt, 2024-07-10:
I just sent over a few dollars/euros to help sponsor your development efforts (by which I mean, "pay for a cold beer"). Vielen dank for all your hard work!

aceisace, 2024-07-10:
Vielen Dank! I was just happy to see a long-time user of Inkycal who is still happy to use it 😉