aceinnolab/Inkycal #368: Waveshare 13,3" error at new cycle

kalalaa, 2024-07-15:
**Waveshare 13,3" error at new cycle**

(venv) inky@inkycal:~/Inkycal $ python inky_run.py
13-07-2024 08:25:35 | inkycal.main |  INFO: Inkycal v2.0.4 booting up...
13-07-2024 08:25:35 | inkycal.main |  INFO: Checking if a settings file is present...
13-07-2024 08:25:35 | inkycal.main |  INFO: Looking for settings.json file in /boot folder...
/home/inky/Inkycal/venv/lib/python3.11/site-packages/gpiozero/devices.py:295: PinFactoryFallback: Falling back from lgpio: module 'lgpio' has no attribute 'SET_BIAS_DISABLE'
  warnings.warn(
13-07-2024 08:25:36 | inkycal.custom.openweathermap_wrapper |  INFO: OWM wrapper initialized for API version 2.5, language en and timezone Europe/Budapest.
13-07-2024 08:25:36 | inkycal.main |  INFO: name : Weather size : 680x141 px
13-07-2024 08:25:36 | inkycal.main |  INFO: name : Calendar size : 680x423 px
13-07-2024 08:25:36 | inkycal.main |  INFO: name : Agenda size : 680x376 px
13-07-2024 08:25:36 | inkycal.main |  INFO: Inkycal initialised successfully!
13-07-2024 08:25:36 | inkycal.main |  INFO: Inkycal version: v2.0.4
13-07-2024 08:25:36 | inkycal.main |  INFO: Selected E-paper display: epd_13_in_3_colour
13-07-2024 08:25:36 | inkycal.main |  INFO: Starting new cycle...
13-07-2024 08:25:36 | inkycal.main |  INFO: Timestamp: 08:25:36 13.07.2024
13-07-2024 08:25:42 | inkycal.modules.ical_parser |  INFO: loaded iCalendars from URLs
13-07-2024 08:25:42 | inkycal.modules.inkycal_agenda |  INFO: Managed to parse events from urls
13-07-2024 08:25:42 | inkycal.main |  INFO: All images generated successfully!
13-07-2024 08:25:43 | inkycal.main |  INFO: Attempting to render image on display...
Initialising..Updating display......Done
Sending E-Paper to deep sleep...Done
13-07-2024 08:26:10 | inkycal.main |  INFO: No errors since 1 display updates
13-07-2024 08:26:10 | inkycal.main |  INFO: program started 35 seconds ago
34 minutes left until next refresh
13-07-2024 09:00:00 | inkycal.main |  INFO: Starting new cycle...
13-07-2024 09:00:00 | inkycal.main |  INFO: Timestamp: 09:00:00 13.07.2024
13-07-2024 09:00:16 | inkycal.modules.ical_parser |  INFO: loaded iCalendars from URLs
13-07-2024 09:00:16 | inkycal.modules.inkycal_agenda |  INFO: Managed to parse events from urls
13-07-2024 09:00:16 | inkycal.main |  INFO: All images generated successfully!
13-07-2024 09:00:17 | inkycal.main |  INFO: Attempting to render image on display...
Initialising..Traceback (most recent call last):
  File "/home/inky/Inkycal/inky_run.py", line 43, in <module>
    asyncio.run(run())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/inky/Inkycal/inky_run.py", line 20, in run
    await inky.run()  # If there were no issues, you can run Inkycal nonstop
    ^^^^^^^^^^^^^^^^
  File "/home/inky/Inkycal/inkycal/main.py", line 368, in run
    display.render(im_black, im_colour)
  File "/home/inky/Inkycal/inkycal/display/display.py", line 94, in render
    epaper.init()
  File "/home/inky/Inkycal/inkycal/display/drivers/epd_13_in_3_colour.py", line 110, in init
    self.send_command(0x12)  # SWRESET
    ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/inky/Inkycal/inkycal/display/drivers/epd_13_in_3_colour.py", line 69, in send_command
    epdconfig.spi_writebyte([command])
  File "/home/inky/Inkycal/inkycal/display/drivers/epdconfig.py", line 94, in spi_writebyte
    self.SPI.writebytes(data)
OSError: [Errno 9] Bad file descriptor
(venv) inky@inkycal:~/Inkycal $

aceisace, 2024-07-16:
Thanks for reporting this issue @kalalaa . It seems there is an issue with the driver file of the 13.3" display. This driver was copied from waveshare directly, but I'll a look into this at the end of today

aceisace, 2024-07-29:
@kalalaa Hi there, apologies for getting back at this rather late. Have you tested this with the latest version of Inkycal? From the looks of it i.e. display updating once, but not after that, this was an issue with the earlier version of InkycalOS-Lite v2.0.4, which was corrected a few days ago. 

Please let me know if it works for you. If not, please paste the full traceback as you have before. Thank you in advance

kalalaa, 2024-07-30:
Hi! The issue is still the same with the latest OS:

inky@inkycal:~ $ cd ~/Inkycal
source venv/bin/activate
git pull
python inky_run.py
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint: 
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint: 
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
Already up to date.
30-07-2024 11:48:32 | inkycal.main |  INFO: Inkycal v2.0.4 booting up...
30-07-2024 11:48:32 | inkycal.main |  INFO: Checking if a settings file is present...
30-07-2024 11:48:32 | inkycal.main |  INFO: Found settings.json file in /boot/settings.json
/home/inky/Inkycal/venv/lib/python3.9/site-packages/gpiozero/devices.py:295: PinFactoryFallback: Falling back from lgpio: module 'lgpio' has no attribute 'SET_BIAS_DISABLE'
  warnings.warn(
30-07-2024 11:48:32 | inkycal.custom.openweathermap_wrapper |  INFO: OWM wrapper initialized for API version 2.5, language hu and timezone Europe/Budapest.
30-07-2024 11:48:32 | inkycal.main |  INFO: name : Weather size : 680x100 px
30-07-2024 11:48:32 | inkycal.main |  INFO: name : Calendar size : 680x376 px
30-07-2024 11:48:32 | inkycal.main |  INFO: name : Agenda size : 680x464 px
30-07-2024 11:48:32 | inkycal.main |  INFO: Inkycal initialised successfully!
30-07-2024 11:48:32 | inkycal.main |  INFO: Inkycal version: v2.0.4
30-07-2024 11:48:32 | inkycal.main |  INFO: Selected E-paper display: epd_13_in_3_colour
30-07-2024 11:48:32 | inkycal.main |  INFO: Starting new cycle...
30-07-2024 11:48:32 | inkycal.main |  INFO: Timestamp: 11:48:32 30.07.2024
30-07-2024 11:48:35 | inkycal.modules.ical_parser |  INFO: loaded iCalendars from URLs
30-07-2024 11:48:38 | inkycal.modules.ical_parser |  INFO: loaded iCalendars from URLs
30-07-2024 11:48:38 | inkycal.modules.inkycal_agenda |  INFO: Managed to parse events from urls
30-07-2024 11:48:39 | inkycal.main |  INFO: All images generated successfully!
30-07-2024 11:48:40 | inkycal.main |  INFO: Attempting to render image on display...
Initialising..Updating display......Done
Sending E-Paper to deep sleep...Done
30-07-2024 11:49:08 | inkycal.main |  INFO: No errors since 55 display updates
30-07-2024 11:49:08 | inkycal.main |  INFO: program started 36 seconds ago
2 minutes left until next refresh
30-07-2024 11:51:00 | inkycal.main |  INFO: Starting new cycle...
30-07-2024 11:51:00 | inkycal.main |  INFO: Timestamp: 11:51:00 30.07.2024
30-07-2024 11:51:03 | inkycal.modules.ical_parser |  INFO: loaded iCalendars from URLs
30-07-2024 11:51:06 | inkycal.modules.ical_parser |  INFO: loaded iCalendars from URLs
30-07-2024 11:51:06 | inkycal.modules.inkycal_agenda |  INFO: Managed to parse events from urls
30-07-2024 11:51:06 | inkycal.main |  INFO: All images generated successfully!
30-07-2024 11:51:07 | inkycal.main |  INFO: Attempting to render image on display...
Initialising..Traceback (most recent call last):
  File "/home/inky/Inkycal/inky_run.py", line 43, in <module>
    asyncio.run(run())
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/inky/Inkycal/inky_run.py", line 20, in run
    await inky.run()  # If there were no issues, you can run Inkycal nonstop
  File "/home/inky/Inkycal/inkycal/main.py", line 372, in run
    display.render(im_black, im_colour)
  File "/home/inky/Inkycal/inkycal/display/display.py", line 94, in render
    epaper.init()
  File "/home/inky/Inkycal/inkycal/display/drivers/epd_13_in_3_colour.py", line 110, in init
    self.send_command(0x12)  # SWRESET
  File "/home/inky/Inkycal/inkycal/display/drivers/epd_13_in_3_colour.py", line 69, in send_command
    epdconfig.spi_writebyte([command])
  File "/home/inky/Inkycal/inkycal/display/drivers/epdconfig.py", line 94, in spi_writebyte
    self.SPI.writebytes(data)
OSError: [Errno 9] Bad file descriptor
(venv) inky@inkycal:~/Inkycal $

Git-Noob-74452, 2025-01-19:
Hi, I have the smae display as mentioned and I am running into the same error. Is there any fix for this? Thanks in advance!

kalalaa, 2025-01-19:
My only fix was this
(crontab -l ; echo "@hourly cd $HOME/Inkycal && venv/bin/python inky_run.py &")| crontab
But i think i set it every 4th hour...
So every 4th hour it starts, refresh the display and exits with the error. But the next 4th hour it starts again...

aceisace, 2025-01-27:
I've got a fix ready for this, but unfortunately no way to test it myself as I do not have the exact same display.
Please run the following to test the fix:

```sh
cd Inkycal
source venv/bin/activate
git pull
git checkout bug/#368
python inky_run.py
# Check if two updates can run one after another
```

Please let me know if this works for you or if it doesn't work as expected.

Git-Noob-74452, 2025-01-27:
Hi thanks for taking a look, but now it will not start at all:

nkycal@calender:~ $ cd $HOME/Inkycal
inkycal@calender:~/Inkycal $ source venv/bin/activate
(venv) inkycal@calender:~/Inkycal $ git pull
remote: Enumerating objects: 25, done.
remote: Counting objects: 100% (25/25), done.
remote: Compressing objects: 100% (15/15), done.
remote: Total 25 (delta 9), reused 19 (delta 9), pack-reused 0 (from 0)
Unpacking objects: 100% (25/25), 12.66 MiB | 2.09 MiB/s, done.
From https://github.com/aceinnolab/Inkycal
   5487464..9a97785  main       -> origin/main
   a701c41..f8f50ef  assets     -> origin/assets
 * [new branch]      bug/#368   -> origin/bug/#368
Updating 5487464..9a97785
Fast-forward
 .github/workflows/update-os.yml | 30 +++++++++++++++++++++++++++---
 requirements.txt                |  6 +++---
 2 files changed, 30 insertions(+), 6 deletions(-)
(venv) inkycal@calender:~/Inkycal $ git checkout bug/#368
branch 'bug/#368' set up to track 'origin/bug/#368'.
Switched to a new branch 'bug/#368'
(venv) inkycal@calender:~/Inkycal $ python3 inky_run.py
27-01-2025 08:05:46 | inkycal.main |  INFO: Inkycal v2.0.4 booting up...
27-01-2025 08:05:46 | inkycal.main |  INFO: Checking if a settings file is present...
27-01-2025 08:05:46 | inkycal.main |  INFO: Found settings.json file in /boot/firmware/settings.json
27-01-2025 08:05:46 | inkycal.custom.openweathermap_wrapper |  INFO: OWM wrapper initialized for API version 2.5, language de and timezone Europe/Berlin.
27-01-2025 08:05:46 | inkycal.main |  INFO: name : Weather size : 680x155 px
27-01-2025 08:05:46 | inkycal.main |  INFO: name : Calendar size : 680x360 px
27-01-2025 08:05:46 | inkycal.main |  INFO: name : Agenda size : 680x425 px
27-01-2025 08:05:46 | inkycal.main |  INFO: Inkycal initialised successfully!
27-01-2025 08:05:46 | inkycal.main |  INFO: Inkycal version: v2.0.4
27-01-2025 08:05:46 | inkycal.main |  INFO: Selected E-paper display: epd_13_in_3_colour
27-01-2025 08:05:46 | inkycal.main |  INFO: Starting new cycle...
27-01-2025 08:05:46 | inkycal.main |  INFO: Timestamp: 08:05:46 27.01.2025
27-01-2025 08:05:53 | inkycal.modules.ical_parser |  INFO: loaded iCalendars from URLs
27-01-2025 08:05:55 | inkycal.modules.inkycal_agenda |  INFO: Managed to parse events from urls
27-01-2025 08:05:56 | inkycal.main |  INFO: All images generated successfully!
27-01-2025 08:05:57 | inkycal.main |  INFO: Attempting to render image on display...
Initialising..Traceback (most recent call last):
  File "/home/inkycal/Inkycal/inky_run.py", line 43, in <module>
    asyncio.run(run())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/inkycal/Inkycal/inky_run.py", line 20, in run
    await inky.run()  # If there were no issues, you can run Inkycal nonstop
    ^^^^^^^^^^^^^^^^
  File "/home/inkycal/Inkycal/inkycal/main.py", line 381, in run
    display.render(im_black, im_colour)
  File "/home/inkycal/Inkycal/inkycal/display/display.py", line 94, in render
    epaper.init()
  File "/home/inkycal/Inkycal/inkycal/display/drivers/epd_13_in_3_colour.py", line 113, in init
    self.send_command(0x12)  # SWRESET
    ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/inkycal/Inkycal/inkycal/display/drivers/epd_13_in_3_colour.py", line 69, in send_command
    epdconfig.spi_writebyte([command])
  File "/home/inkycal/Inkycal/inkycal/display/drivers/epdconfig.py", line 94, in spi_writebyte
    self.SPI.writebytes(data)
OSError: [Errno 9] Bad file descriptor
(venv) inkycal@calender:~/Inkycal $

aceisace, 2025-01-29:
Thanks for testing this out @Git-Noob-74452 . Will take a deeper look at this on the weekend 👍

closed 2025-04-07 by github-actions[bot]