aceinnolab/Inkycal #369: Agenda will not be updated

GimmeMe1, 2024-08-20:
Hello
Since a few days, the agenda (modul 3) is not updated anymore. 
The inkycal.log stated some errors and i have no idea how to fix this unfortunatly:
20-08-2024 15:00:00 | inkycal.main |  INFO: Starting new cycle...
20-08-2024 15:00:00 | inkycal.main |  INFO: Timestamp: 15:00:00 20.08.2024
20-08-2024 15:00:01 | inkycal.custom.openweathermap_wrapper |  INFO: OpenWeatherMap response did not contain a wind gus>
20-08-2024 15:00:13 | inkycal.modules.ical_parser |  INFO: loaded iCalendars from URLs
20-08-2024 15:00:14 | inkycal.main |  ERROR: Error in module 3!
Traceback (most recent call last):
  File "/home/inky/Inkycal/inkycal/main.py", line 611, in process_module
    black, colour = module.generate_image()
                    ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/inky/Inkycal/inkycal/modules/inkycal_agenda.py", line 130, in generate_image
    upcoming_events = parser.get_events(today, agenda_events[-1]['begin'],
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/inky/Inkycal/inkycal/modules/ical_parser.py", line 133, in get_events
    if events: self.parsed_events += list(events)
                                     ^^^^^^^^^^^^
  File "/home/inky/Inkycal/inkycal/modules/ical_parser.py", line 123, in <genexpr>
    arrow.get(events.get('dtstart').dt).format('HH:mm') != '00:00')
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/inky/Inkycal/venv/lib/python3.11/site-packages/arrow/api.py", line 91, in get
    return _factory.get(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/inky/Inkycal/venv/lib/python3.11/site-packages/arrow/factory.py", line 242, in get
    return self.type.fromdatetime(arg, tzinfo=tz)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/inky/Inkycal/venv/lib/python3.11/site-packages/arrow/arrow.py", line 329, in fromdatetime
    return cls(
           ^^^^
  File "/home/inky/Inkycal/venv/lib/python3.11/site-packages/arrow/arrow.py", line 173, in __init__
    tzinfo = parser.TzinfoParser.parse(tzinfo.zone)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/inky/Inkycal/venv/lib/python3.11/site-packages/arrow/parser.py", line 769, in parse
    raise ParserError(f"Could not parse timezone expression {tzinfo_string!r}.")
arrow.parser.ParserError: Could not parse timezone expression 'Customized Time Zone'.
20-08-2024 15:00:15 | inkycal.main |  INFO: Attempting to render image on display...
20-08-2024 15:00:45 | inkycal.main |  INFO: No errors since 0 display updates
20-08-2024 15:00:45 | inkycal.main |  INFO: program started 16 hours ago

Thank you and best Thomas

aceisace, 2024-08-20:
HI there @GimmeMe1 . As the code of the agenda module has not been changed for quite some time now, it's safe to assume the issue lies elsewhere. 

The error you are getting is related to the fact that the timezone cannot be parsed correctly. This could be either due to the iCalendar itself, some recently added event which may be related to timezones or a config issue. 

Could you provide some more details to help debug this issue, for example, what timezone was set on the Raspberry Pi via raspi-config? Apart from this, the section containing the config for the agenda module with sensitive data removed would help reproduce this issue

GimmeMe1, 2024-08-20:
Hello @aceisace 
Yep, a new entry might be the reason. I recently added some entries and the problem might be occur since then but i have no idea what the specific problem with these entries can be. There are some series as well.

Timezone at raspi is Zurich/Europe
Current default time zone: 'Europe/Zurich'
Local time is now:      Tue Aug 20 22:53:51 CEST 2024.
Universal Time is now:  Tue Aug 20 20:53:51 UTC 2024.

Config of the module is

        {
            "position": 3,
            "name": "Agenda",
            "config": {
                "size": [
                    480,
                    390
                ],
                "ical_urls": "https://sharing.calendar.live.com/calendar/private/xxxxx/9>
                "ical_files": null,
                "date_format": "ddd D MMM",
                "time_format": "HH: mm",
                "padding_x": 0,
                "padding_y": 0,
                "fontsize": 16,
                "language": "de"
            }
        }
    ],
    "language": "de",
    "calibration_hours": [
        0,
        12
    ]

Best Thomas

GimmeMe1, 2024-08-20:
some entries have for no specific reason time zone +1 Sarajevo, Warschau, Zagreb. Maybe this is the problem

GimmeMe1, 2024-08-21:
time zone +1 Sarajevo, Warschau, Zagreb seems not to be the problem...

GimmeMe1, 2024-08-21:
the problem is the "customized time zone" entry, I don't know how and why it was set in the calendar. Since I changed all of these entries the agenda display is working fine again. Thanks

SUMMARY:Vxxxxxxx
DTSTART;TZID=Customized Time Zone:20240618T080000
DTEND;TZID=Customized Time Zone:20240618T120000
CLASS:PUBLIC
PRIORITY:5

aceisace, 2024-08-22:
I'm glad to hear your issue was resolved, Thomas. Yes, it seems the  `Customized Time Zone` was indeed the issue. I'll be closing this issue as it has been resolved

GimmeMe1, 2024-08-22:
perfect!