aceinnolab/Inkycal #377: [FEATURE]: match formatting of calendar & agenda events, and make calendar events optional

ounrequitedo, 2024-09-17:
**Is your feature request related to a problem? Please describe.**
when adding the calendar _**and**_ agenda modules, you can easily see the difference in formatting of the two modules, which are kind of redundant in function

**Describe the solution you'd like**
please remove agenda items section from the calendar module or make them optional with a toggle, as it's already served by the exclusive agenda module anyways and the formatting of both do not match (the agenda module is nicer and easier to read too imo) 

**Describe alternatives you've considered**
in the meantime, working directly with @aceisace i've hidden the events section of the Calendar app by changing the code in the calendar modules, but it may not persist through upgrades 

**Additional context**
thanks for the discord help!

git-mAdHeN, 2025-03-27:
Hi Ounrequitedo,

i also think that the agenda entries under the calendar are unnecessary. Nevertheless, I think it's good when the days with appointments are marked in the calendar. Can you tell me how you changed the code?

ounrequitedo, 2025-03-27:
> Hi Ounrequitedo,
> 
> i also think that the agenda entries under the calendar are unnecessary. Nevertheless, I think it's good when the days with appointments are marked in the calendar. Can you tell me how you changed the code?

from my chat with the dev on discord (super friendly and helpful):
[image]

_While there isn't yet a separation between mark-day-with-events and display-events below calendar in the calendar module, with some hacking, you can easily get this done by modifying a few lines:

First, open the calendar module file on SSH: nano ~/Inkycal/inkycal/modules/inkycal_calendar.py

Then, modify the lines so that they look like the version on the right of both images. Left side is before the change, as they should be in your file at the moment, the right side is the updated file._

also:
_To hide the dashed line, you can remove the following section, right at the bottom of the file:_
`            else:
                symbol = '- '

                while self.font.getlength(symbol) < im_width * 0.9:
                    symbol += ' -'
                write(
                    im_black,
                    event_lines[0],
                    (im_width, line_height),
                    symbol,
                    font=self.font,
                )`

git-mAdHeN, 2025-03-27:
Thank you! I appreciate your help.

closed 2026-02-22 by github-actions[bot]