aceinnolab/Inkycal #378: [FEATURE]: Battery level on PiSugar version please

ounrequitedo, 2024-09-17:
**Is your feature request related to a problem? Please describe.**
don't know battery level

**Describe the solution you'd like**
Module for a single-line battery-level % or better yet addition to the debug line at the bottom showing last update

**Describe alternatives you've considered**
installing i2c-tools, i couldn't get the battery level to read out and gave up

ounrequitedo, 2024-09-20:
bump because i was asked to ^_^

aceisace, 2024-09-20:
Thanks for bumping this up. I've implemented this feature in a feature branch, which you can test like this:
```sh
cd ~/Inkycal
source venv/bin/activate
git stash # in case you have custom changes
git pull # update the repo
git stash pop # undo the custom changes
git checkout feature/#378 # switch to branch with this feature request
python inky_run.py
```

It basically boils down to the following:
[image]

Please let me know if this works for you so I can merge the PR 👍

closed 2024-09-22 by aceisace