aceinnolab/Inkycal #379: Error 500 on web UI

g0emf, 2024-09-20:
When in the Web UI I  "get Ohoh, this wasn't expected (Error 500)
Apologies for the inconvenience!"

**To Reproduce**
Go to the Web UI to generate the settings.json file
Select the display 5.83 B/W V2 
add weather & joke modules
Click Generate file

**Expected behavior**
This should generate the settings file

**Desktop (please complete the following information):**
 Windows 11 64 bit
 - Browser  chrome, brave firefox
 
Note other models of displays works without problem (Generate the settings.json file)
As I only have the 5.83 V2 Black & White display I am not able to test further

[image]

aceisace, 2024-09-20:
@g0emf Thanks for reporting this issue!
I've implemented the bugfix for this. Please try it again now and it should hopefully work. 
I'll be closing this issue as it seems to work on my end, however, if it doesn't work as expected, please feel free to re-open this issue.

closed 2024-09-20 by aceisace

g0emf, 2024-09-21:
Yes that's now working thanks so much for the fix
Best regards from the UK
Dave

Off to have a play

aceisace, 2024-09-21:
Glad to hear it's fixed now, off you go and thanks for sponsoring!

g0emf, 2024-09-21:
The least I can do, it is very easy now it's working!

I have just been playing about trying to get my local tide information, The best I can do is capture a jpeg from the web... not ideal but an excellent project for my e-paper.
Thanks again Ace best regards
Dave

aceisace, 2024-09-23:
Thanks very much then 😁
Tide-information sounds interesting too! Are there APIs for this too?

g0emf, 2024-09-23:
Yes this one you can use for free 
https://www.admiralty.co.uk/access-data/apis
Yahoo Mail: Search, organise, conquer

aceisace, 2024-09-23:
Too bad they require a subscription to access the data, but nice to see there are APIs for this too!

g0emf, 2024-09-23:
Hi Ace,

I jumped into Admiralty and got a free dev API but it's only for the UK so not suitable for Inkycal.

Just had a Google and found https://stormglass.io/pricing/ its free, 10 requests per day we only need 1.
Worth a look?

Thanks for a really good project.

Dave