aceinnolab/Inkycal #381: WaveShare 10.3inch e-Paper HAT Issue

lbrent2k, 2024-09-29:
Nothing displays - see output provided below.

Per WaveShare Wiki VCOM should be: epd -1.50 1 though my display indicates 1.46, both work.

Can I modify configuration to point to: /home/lbrent/bcm2835-1.60/IT8951 which I installed from WaveShare? When I run demo from WaveShare, the display works.

 python3 inky_run.py
29-09-2024 09:43:25 | inkycal.main |  INFO: Inkycal v2.0.4 booting up...
29-09-2024 09:43:25 | inkycal.main |  INFO: Checking if a settings file is present...
29-09-2024 09:43:25 | inkycal.main |  INFO: Found settings.json file in /boot/firmware/settings.json
29-09-2024 09:43:25 | inkycal.custom.openweathermap_wrapper |  INFO: OWM wrapper initialized for API version 2.5, language en and timezone US/Eastern.
29-09-2024 09:43:25 | inkycal.main |  INFO: name : Weather size : 1404x926 px
29-09-2024 09:43:25 | inkycal.main |  INFO: Inkycal initialised successfully!
29-09-2024 09:43:25 | inkycal.main |  INFO: Inkycal version: v2.0.4
29-09-2024 09:43:25 | inkycal.main |  INFO: Selected E-paper display: 10_in_3
29-09-2024 09:43:25 | inkycal.main |  INFO: Starting new cycle...
29-09-2024 09:43:25 | inkycal.main |  INFO: Timestamp: 09:43:25 29.09.2024
29-09-2024 09:43:25 | inkycal.modules.inkycal_weather |  INFO: Please consider decreasing the height.
29-09-2024 09:43:26 | inkycal.custom.openweathermap_wrapper |  INFO: OpenWeatherMap response did not contain a wind gust speed. Using base wind: 2 m/s.
29-09-2024 09:43:27 | inkycal.main |  INFO: All images generated successfully!
29-09-2024 09:43:28 | inkycal.main |  INFO: Attempting to render image on display...
29-09-2024 09:43:28 | inkycal.modules.inky_image |  INFO: merged given images into one
Initialising..Updating display......sudo /home/lbrent/Inkycal/inkycal/display/drivers/parallel_drivers/epd -2.0 0 /home/lbrent/Inkycal/inkycal/../image_folder/canvas.bmp
/***********************************/ 
bcm2835 init success !!! 
/***********************************/ 
VCOM value:2000
Display mode:0
Panel(W,H) = (1872,1404)
Memory Address = 119F00
FW Version = SWv_0.1.
LUT Version = M841_TFA5210
VCOM = -2.00V
A2 Mode:6
Done
Sending E-Paper to deep sleep...Done
29-09-2024 09:43:30 | inkycal.main |  INFO: No errors since 6 display updates
29-09-2024 09:43:30 | inkycal.main |  INFO: program started just now
7 minutes left until next refresh

aceisace, 2024-09-29:
Hi there @lbrent2k and thanks for reporting this issue. 
Inkycal was able to interface with the display, proving the correct wiring and driver selection. As the waveshare demo also works with the same setup, there's a high chance the issue is related to the settings.json file. 

More particularly, the weather module doesn't like much height as it will attempt to stretch the height to the given height, resulting in some distortions. Please try this again with the jokes module or use several modules and adapt at least the height of the weather module to around 20. 

The new settings.json file should override the current one and the Raspberry Pi needs to restart after this, which will hopefully fix the issue.

Please let me know if that worked

lbrent2k, 2024-09-29:
I modified the module to jokes.  Still, no screen refresh, no screen output.
python3 inky_run.py
29-09-2024 11:01:30 | inkycal.main |  INFO: Inkycal v2.0.4 booting up...
29-09-2024 11:01:30 | inkycal.main |  INFO: Checking if a settings file is present...
29-09-2024 11:01:30 | inkycal.main |  INFO: Found settings.json file in /boot/firmware/settings.json
29-09-2024 11:01:30 | inkycal.main |  INFO: name : Jokes size : 1404x1852 px
29-09-2024 11:01:30 | inkycal.main |  INFO: Inkycal initialised successfully!
29-09-2024 11:01:30 | inkycal.main |  INFO: Inkycal version: v2.0.4
29-09-2024 11:01:30 | inkycal.main |  INFO: Selected E-paper display: 10_in_3
29-09-2024 11:01:30 | inkycal.main |  INFO: Starting new cycle...
29-09-2024 11:01:30 | inkycal.main |  INFO: Timestamp: 11:01:30 29.09.2024
29-09-2024 11:01:30 | inkycal.main |  INFO: All images generated successfully!
29-09-2024 11:01:31 | inkycal.main |  INFO: Attempting to render image on display...
29-09-2024 11:01:32 | inkycal.modules.inky_image |  INFO: merged given images into one
Initialising..Updating display......sudo /home/lbrent/Inkycal/inkycal/display/drivers/parallel_drivers/epd -2.0 0 /home/lbrent/Inkycal/inkycal/../image_folder/canvas.bmp
/***********************************/ 
bcm2835 init success !!! 
/***********************************/ 
VCOM value:2000
Display mode:0
Panel(W,H) = (1872,1404)
Memory Address = 119F00
FW Version = SWv_0.1.
LUT Version = M841_TFA5210
VCOM = -2.00V
A2 Mode:6
Done
Sending E-Paper to deep sleep...Done
29-09-2024 11:01:34 | inkycal.main |  INFO: No errors since 7 display updates
29-09-2024 11:01:34 | inkycal.main |  INFO: program started just now
9 minutes left until next refresh

aceisace, 2024-09-29:
I just realised there are now two different versions of the 10.3" e-paper hat. Which one did you buy, the normal one or the 10.3inch e-Paper HAT (D) one?

lbrent2k, 2024-09-29:
I am using this: https://www.waveshare.com/10.3inch-e-paper-hat.htm It includes the driver board.

lbrent2k, 2024-10-01:
I think this should be changed to Bug instead of Question since I have verified that display works when running the WaveShare demo and it appears that inkycal has been installed and configured properly, yet the display remains unresponsive

aceisace, 2024-10-01:
May I kindly ask you which commands you had to run to test the waveshare code? At the moment, I am not sure what is different from the waveshare-version.

lbrent2k, 2024-10-01:
I am following directions from: https://www.waveshare.com/wiki/10.3inch_e-Paper_HAT to succesfully run the WaveShare Demo:

cd bcm2835-1.60
./configure
make
sudo make check
sudo make install
sudo apt install gpiod libgpiod-dev
cd IT8951/
sudo make clean
sudo make -j4
sudo ./epd -1.50 1

closed 2025-04-13 by github-actions[bot]