aceinnolab/Inkycal #385: Compatibility with Inky Impression

kitjohnson, 2024-12-31:
Hi there,

Would this project be compatible with the Inky Impression?

https://shop.pimoroni.com/products/inky-impression-7-3?variant=40512683376723

I can see there is a related Python library for Inky devices - would this be cross-compatible?

https://github.com/pimoroni/inky

Thank you in advance!

aceisace, 2024-12-31:
Hi there and thanks for your interest.
Adding support for pimoroni displays is a fairly large task, especially since adding support for more than three colours is already a challenge in Inkycal. At this moment of time at least, it would be too much work on a feature used by very few people who have this display. For now, I cannot say if and when I can work on this or if it is possible.

kitjohnson, 2025-01-01:
Thank you Ace!