aceinnolab/Inkycal #387: Potential fix for the `Failed to add edge detection` error using the latest version of the Raspberry OS ?

pbarthelemy, 2025-01-18:
Hello,

As mentioned in #help on Discord, I ran into the known issue about the failed edge detection, using the latest version of the Raspberry OS installed via the Raspberry imager default.
I used ~a few google searches with ensuing cut-and-paste~ my awesome brain power to solve the issue...

I am really not a linux/gpio expert :  the steps below worked for me, I can't go beyond that... 
Maybe it is a reliable fix/workaround : throwing it to this community for tests and comment !

I copied the style of the main doc to document the fix. 

happy Inkycaling !
--Philippe

## `Failed to add edge detection` error : using the latest version of the Raspberry OS.

Using the latest version of the Raspberry OS can lead to the "Failed to add edge detection" error when trying to run InkyCal.
(The latest version of the OS is the one you will get by default from the Raspberry Imager application)
This is a known Waveshare library issue, which can not be solved in the InkyCal code itself.
The error will trigger the error message below.

```
  File "/home/johndoe/Inkycal/venv/lib/python3.11/site-packages/gpiozero/pins/pi.py", line 636, in _set_when_changed
    self._enable_event_detect()
  File "/home/john/Inkycal/venv/lib/python3.11/site-packages/gpiozero/pins/rpigpio.py", line 220, in _enable_event_detect
    GPIO.add_event_detect(
RuntimeError: Failed to add edge detection
```

### What can I do to fix the error ?
- Reinstall an older version of the Raspberry OS ([bullseye from 2023-05-03](https://downloads.raspberrypi.com/raspios_lite_armhf/images/raspios_lite_armhf-2023-05-03/2023-05-03-raspios-bullseye-armhf-lite.img.xz) ) as noted at the top of the [configuring-the-raspberry-pi](https://github.com/aceinnolab/Inkycal?tab=readme-ov-file#configuring-the-raspberry-pi) section.

- Keep the latest OS and use a GPIO library called lgpio (l for Legacy) which manages the issue

### I installing LGPIO instead for the GPIO
If you choose the second option, expect a simple but somewhat lengthy installation.
The steps are detailed below :

```bash
# follow the installation procedure up to (included) "pip install RPi.GPIO==0.7.1 spidev==3.5 gpiozero==2.0"
# do execute the pip install above

# first, let's uninstall the GPIO library
 pip uninstall RPi.GPIO

# the other gpio library requires three libraries, let's install them
wget  https://github.com/PCRE2Project/pcre2/releases/download/pcre2-10.44/pcre2-10.44.tar.bz2
bzip2 -d pcre2-10.44.tar.bz2
tar -xf pcre2-10.44.tar
cd pcre2-10.44/
./configure && make && sudo make install && make clean
# grab a coffee... or two ...
cd ..

wget https://sourceforge.net/projects/swig/files/swig/swig-4.3.0/swig-4.3.0.tar.gz
tar zxvf swig-4.3.0.tar.gz
cd swig-4.3.0/
./configure && make && sudo make install && make clean
# enough coffee, go for a walk ...
cd ..

wget http://abyz.me.uk/lg/lg.zip
unzip lg.zip
cd lg
make && sudo make install && make clean
cd ..

# and we are ready to install the other GPIO library
cd Inkycal/
pip install rpi-lgpio

# and there we go 
python3 inky_run.py
```

aceisace, 2025-01-26:
Finally tested this through, I have to say it takes several hours from start to end to get this up and running, therefore I won't be adding these instructions as the main process, but as a special section under the wiki and integrate these steps into InkycalOS-Lite. Would you like to get access to the new version of InkycalOS-Lite to test if everything works as expected?

On my end it works and I honestly can't thank you enough!

pbarthelemy, 2025-01-26:
> Finally tested this through, I have to say it takes several hours from start to end to get this up and running, therefore I won't be adding these instructions as the main process, but as a special [...]

You are welcome !
100% agree with your decision  : could be be just too loooooong and also simply not warranted for 99.9% of the users.

I would happy to get access to the latest of InkycalOS-Lite and test it... just in case... (caveat emptor: I work in a QA dept.  😁)
(can't commit on a timeframe for this quick test... busy on the family-side of things... do not wait for me ) 

--p

aceisace, 2025-01-27:
Thanks very much! I've just finished integrating it into the new version of InkycalOS-Lite, hence it's ready for testing.

It's okay if you're busy and I respect that. Hence I'll take care of testing it on my end and will send you the link once I've finished testing the new version.
On that note, please send me a DM on Discord with your email-address since I need it to send the link to once it's ready.

closed 2025-04-06 by github-actions[bot]