aceinnolab/Inkycal #388: 'Command not found' error, no output to eink display

Napmaster3000, 2025-01-26:
When launching the following error shows:

Initialising..Updating display......sudo /home/pi/Inkycal/inkycal/display/drivers/parallel_drivers/epd -2.0 0 /home/pi/Inkycal/inkycal/../image_folder/canvas.bmp
sudo: /home/pi/Inkycal/inkycal/display/drivers/parallel_drivers/epd: command not found

No output is shown on the screen.

Repro steps:

Install per instructions on fresh Bookworm 64bit Lite OS

Run source venv/bin/activate
python3 inky_run.py

I have formatted SD and tried again from scratch and still see the same error

aceisace, 2025-01-26:
HI there and thanks for reporting this issue. Can you provide some more details about your setup?;
- Which display version are you using?
- Which username did you set when flashing the OS?
- Have you tried this with the 32-bit version too?
- If possible, a copy of your settings.json file with any sensitive data removed

Thanks in advance. I'll get back to this once I have some more information 👍

Napmaster3000, 2025-01-26:
Hi there, thanks or the response.

The display is the 7.8 inch found here:
https://www.waveshare.com/7.8inch-e-paper-hat.htm

Username is pi

I have not yet tried the 32bit OS

Json file is here:

{
    "model": "7_in_8",
    "update_interval": 15,
    "orientation": 0,
    "info_section": true,
    "info_section_height": 20,
    "border_around_modules": false,
    "modules": [
        {
            "position": 1,
            "name": "Weather",
            "config": {
                "size": [
                    1404,
                    617
                ],
                "api_key": "XXXXXXXXXXXXXXXXXXXXXX",
                "location": "XXXXXXX",
                "round_temperature": true,
                "round_windspeed": true,
                "forecast_interval": "daily",
                "units": "metric",
                "hour_format": "12",
                "use_beaufort": true,
                "padding_x": 0,
                "padding_y": 0,
                "fontsize": 12,
                "language": "en"
            }
        },
        {
            "position": 2,
            "name": "Agenda",
            "config": {
                "size": [
                    1404,
                    617
                ],
                "ical_urls": "https://ext.todoist.com/export/ical/todoist?user_id=XXXXXXXXXXXXXXXX",
                "ical_files": null,
                "date_format": "ddd D MMM",
                "time_format": "HH: mm",
                "padding_x": 0,
                "padding_y": 0,
                "fontsize": 12,
                "language": "en"
            }
        },
        {
            "position": 3,
            "name": "Todoist",
            "config": {
                "size": [
                    1404,
                    617
                ],
                "api_key": "XXXXXXXXXXXXXXXXX",
                "project_filter": null,
                "padding_x": 0,
                "padding_y": 0,
                "fontsize": 12,
                "language": "en"
            }
        }
    ],
    "language": "en",
    "calibration_hours": [
        0,
        12,
        18
    ]
}

Napmaster3000, 2025-01-26:
UPDATE: I have attempted to install using the recommended image per the wiki, the error is persistent:

pi@inkycal:/boot $ cd $HOME/Inkycal
source venv/bin/activate
python3 inky_run.py
27-01-2025 10:29:16 | inkycal.main |  INFO: Inkycal v2.0.4 booting up...
27-01-2025 10:29:16 | inkycal.main |  INFO: Checking if a settings file is present...
27-01-2025 10:29:16 | inkycal.main |  INFO: Found settings.json file in /boot/settings.json
27-01-2025 10:29:16 | inkycal.custom.openweathermap_wrapper |  INFO: OWM wrapper initialized for API version 2.5, language en and timezone Australia/Melbourne.
27-01-2025 10:29:16 | inkycal.main |  INFO: name : Weather size : 1404x308 px
27-01-2025 10:29:17 | inkycal.main |  INFO: name : Agenda size : 1404x309 px
27-01-2025 10:29:17 | inkycal.main |  INFO: name : Todoist size : 1404x1255 px
27-01-2025 10:29:17 | inkycal.main |  INFO: Inkycal initialised successfully!
27-01-2025 10:29:17 | inkycal.main |  INFO: Inkycal version: v2.0.4
27-01-2025 10:29:17 | inkycal.main |  INFO: Selected E-paper display: 7_in_8
27-01-2025 10:29:17 | inkycal.main |  INFO: Starting new cycle...
27-01-2025 10:29:17 | inkycal.main |  INFO: Timestamp: 10:29:17 27.01.2025
27-01-2025 10:29:24 | inkycal.modules.ical_parser |  INFO: loaded iCalendars from URLs
27-01-2025 10:29:24 | inkycal.modules.inkycal_agenda |  INFO: Managed to parse events from urls
27-01-2025 10:29:25 | inkycal.modules.inkycal_todoist |  INFO: Connection test passed
27-01-2025 10:29:28 | inkycal.main |  INFO: All images generated successfully!
27-01-2025 10:29:38 | inkycal.main |  INFO: Attempting to render image on display...
27-01-2025 10:29:40 | inkycal.modules.inky_image |  INFO: merged given images into one
Initialising..Updating display......sudo /home/pi/Inkycal/inkycal/display/drivers/parallel_drivers/epd -2.0 0 /home/pi/Inkycal/inkycal/../image_folder/canvas.bmp
sudo: unable to execute /home/pi/Inkycal/inkycal/display/drivers/parallel_drivers/epd: No such file or directory
Don

aceisace, 2025-01-29:
@Napmaster3000 This looks like a bug then, will take a look at this on the weekend 👍

Napmaster3000, 2025-02-09:
Hi there, just thought I'd check in to see if you might have made any progress on this bug? Thanks!

closed 2025-04-17 by github-actions[bot]