aceinnolab/Inkycal #390: [FEATURE]: defining borders around the content, rendering Inkycal on an area smaller than the full panel

pbarthelemy, 2025-02-11:
**Is your feature request related to a problem? Please describe.**
Inkycal assumes the full e-ink panel is useable and visible.

However, when put in a standard picture frame, a significant section around the panel can be hidden : for instance, this happens when using a 7.5in panel with 5*7 Ikea photo frame as its cut-out is smaller than the screen. There are certainly other uses cases.

The new feature is about defining a useable area that is smaller than the panel, i.e. defining borders around the content and rendering it at the appropriate position.

**Describe the solution you'd like**

1. add 4 parameters in settings.json to define the widths of the 4 border strips (`frame_border_x1`, etc... in pixels) 
2. modify the web-ui to calculate the module sizes, accounting for the smaller useable area
3. make minimal changes in the Inkycal code to assemble the modules in the useable area
4. add a step to paste the useable area to final full-size image before it is rendered onto the panel

_nota_ : 
- the setting names `frame_border_x1`, `frame_border_x2`, `frame_border_y1`, `frame_border_y2` are generic : portrait or landscape, they remain meaningful in both cases
- there are 4 values as it is not assumed the useable area is centered on panel
- the padding at the module-level is still available.
- this feature can easily made backwards-compatible : if the 4 new settigns are qbsent from settings.json, the border widths default to 0 and the image is rendered as is the current version.

**Describe alternatives you've considered**
Currently, one way to address the problem is to use padding x-axis and padding y-axis. This creates a border in each module.
While it is workable, this solution does not apply to the debug section (and its PiSugar info) and has to implemented in each module.

**Additional context**
This addresses [FEATURE]: image module -> allow max width in percent #354

**Example**
draft image - asymmetric borders shozn in red as an highlight 
`    "frame_x1" : 20,
    "frame_x2" : 40,
    "frame_y1" : 10,
    "frame_y2" : 50,` in the settings.json, on a 800*480 panel
[image]

pbarthelemy, 2025-02-15:
@aceisace : probably a stupid question ;-) ... 
Where is the code for the web-ui ?
(I did not find it in the repo... I guess I am missing something ... )

aceisace, 2025-02-22:
Thanks for the PR! Will get back to this and your PR in a few days. The code for the web-ui is not (yet) public. However, I can adapt the code if any changes are needed 👍

closed 2026-04-17 by github-actions[bot]