aceinnolab/Inkycal #397: Error while installing

HiroP77, 2025-07-22:
while doing a fresh installation, the following shows up

Collecting kiwisolver==1.4.5 (from inkycal==2.0.4)
  ERROR: HTTP error 403 while getting https://archive1.piwheels.org/simple/kiwisolver/kiwisolver-1.4.5-cp39-cp39-linux_armv6l.whl#sha256=a08aab5f2b69b2658efbbd8661e556ff462f72b0601644da36de0d9e55a05a99 (from https://www.piwheels.org/simple/kiwisolver/) (requires-python:>=3.7)
ERROR: Could not install requirement kiwisolver==1.4.5 from https://archive1.piwheels.org/simple/kiwisolver/kiwisolver-1.4.5-cp39-cp39-linux_armv6l.whl#sha256=a08aab5f2b69b2658efbbd8661e556ff462f72b0601644da36de0d9e55a05a99 (from inkycal==2.0.4) because of HTTP error 403 Client Error: Forbidden for url: https://archive1.piwheels.org/simple/kiwisolver/kiwisolver-1.4.5-cp39-cp39-linux_armv6l.whl for URL https://archive1.piwheels.org/simple/kiwisolver/kiwisolver-1.4.5-cp39-cp39-linux_armv6l.whl#sha256=a08aab5f2b69b2658efbbd8661e556ff462f72b0601644da36de0d9e55a05a99 (from https://www.piwheels.org/simple/kiwisolver/) (requires-python:>=3.7)

Starting of Inkycal then leads to
~/Inkycal $ python3 inky_run.py
Traceback (most recent call last):
  File "/home/volker/Inkycal/inky_run.py", line 8, in <module>
    from inkycal import Inkycal
  File "/home/volker/Inkycal/inkycal/__init__.py", line 2, in <module>
    import inkycal.modules.inkycal_agenda
  File "/home/volker/Inkycal/inkycal/modules/__init__.py", line 1, in <module>
    from .inkycal_agenda import Agenda
  File "/home/volker/Inkycal/inkycal/modules/inkycal_agenda.py", line 5, in <module>
    import arrow # noqa
ModuleNotFoundError: No module named 'arrow'

aceisace, 2025-07-29:
@jimusik Thanks for the information about the Pi running out of swap during installation. I'll adapt the README accordingly. 

Thanks for your wait @HiroP77 . Please try out the fix suggested by @jimusik and let me know if this fixed your issue

aceisace, 2025-07-29:
Added to README 👍 
Default recommended swap size has been increased from 512MB to 1024MB

HiroP77, 2025-07-29:
Yes, it seems to have fixed it, thanks for your help, @jimusik

closed 2025-07-29 by HiroP77